ricaun/LoRaNow #1: Gateway não recebe dados

lucasmaziero, 2019-04-05:
Olá,

Estou usando os exemplos de node e gateway, mas não esta funcionando, o gateway não recebe os pacotes de dados.

ricaun, 2019-04-05:
Olá Lucas,
Pode ter vários fatores, suponho que o 'begin' esta iniciando normalmente sem dar erro e o node esta enviando a mensagem normalmente.
As vezes acontece de uma placa esta muito próxima da outra e os pacotes chegam com bastante interferência, ai o gateway ignora o pacote por não passar no check.
Tu pode entrar no 'LoRaNow.h' e habilitar o `#define LORANOW_DEBUG_STREAM Serial` e ver se esta recebendo algum pacote.
Tais utilizando que placa?!

lucasmaziero, 2019-04-05:
Já fiz esse teste de habilitar o debug na serial, mas não recebo nada, usei o exemplo do "LoRa.h" a lib que você usa como base e tudo funciona normalmente, estou usando um Arduino e um RFM95.

ricaun, 2019-04-05:
Que estranho... Estou tentando imaginar o que pode ser.
Testou com o exemplo `LoRaReceiverCallback`? Muito importante o DIO0 tem que estar no pino 2, não é algo opcional.
Qual tu frequência esta utilizando?!

lucasmaziero, 2019-04-05:
Sim o DIO0 está no pino 2 e frequência e 915 MHz como os exemplos "LoRaSimpleGateway" e "LoRaSimpleNode" da lib "LoRa.h" que justamente utiliza interrupções funciona tudo certo, o estranho que quanto uso a "LoRaNow.h" não funciona? Você já testou esses exemplos da lib sem problemas?

ricaun, 2019-04-05:
Tava testando aqui acabei achando um bug. Mas acredito que não esta relacionado a esse teu problema.
Já que não estou resetando o modulo RF, as vezes fica alguma configuração antiga.
Tenho que adicionar essa configuração pra padronizar.
```
LoRa.setSignalBandwidth(125E3);
LoRa.setCodingRate4(5);
LoRa.setPreambleLength(8);
LoRa.setSyncWord(0x12);
LoRa.disableCrc();
```

Mas se o exemplo `LoRaSimpleGateway` e o `LoRaSimpleNode` funciona, não deve ser esse o problema.
Acabei de testar o exemplo do `LoRaSimpleGateway` e outro modulo com o exemplo `LoRaNow_Node`.
Só tive que mudar a no exemplo `LoRaSimpleGateway ` a frequência para 917E6, e a comunicação funcionou, no caso o gateway recebe mensagem.

Não sei o que tem de errado no teu setup, estou achando que o gateway e o node estão muito proximos um do outro, tenta colocar uma distancia de dois metros entre eles.

Me manda o que tu esta recebendo no Serial, quando o `#define LORANOW_DEBUG_STREAM Serial` esta acionado.

ricaun, 2019-04-24:
Modifiquei algumas coisas na biblioteca que tava dando problema,
então se conseguir testar novamente agradeço.

closed 2019-04-24 by ricaun