ricaun/LoRaNow #3: Finallyyy...

halukmy, 2019-04-23:
i been really really serching beautiful library.. and you got you bro..

please do not support this baby..

thanks :)

ricaun, 2019-04-24:
The idea was to make simple as possible and design to work with most used hardware available.
Thanks, I really appreciate that.

closed 2019-04-24 by ricaun

halukmy, 2019-04-24:
@ricaun i just made the old framework for me like gateway id 222 and node id is 200
if (id 222) gateway or if id (200) node

is it possible on this library?

my second question is,

i just tested radio freq. on other arduino library

when i set 915e6 i get 600~ mhz, when i made it 1020, i get around 686 - 700 mhz,..

is your repo fix that problem bro?

thanks

ricaun, 2019-04-24:
> when i set 915e6 i get 600~ mhz, when i made it 1020, i get around 686 - 700 mhz,..

Technically I'm using the [arduino-LoRa](https://github.com/sandeepmistry/arduino-LoRa) library to make all the LoRa configuration stuff. So I never have a problem with frequency as you explain.

> @ricaun i just made the old framework for me like gateway id 222 and node id is 200
> if (id 222) gateway or if id (200) node

I didn't understand want you mean, but each node have a UniqueID and the gateway always listens for data, if it's found a valid protocol the gateway can read packege and send back a data to the node id. Similar as LoRaWan Class A.