ricaun/LoRaNow #4: ttgo lora 32 got problems

halukmy, 2019-04-24:
i just set   LoRa.setPins(18, 26); // set CS, reset, IRQ pin

.
Connected to sungerbob
IP address: 192.168.0.25
HTTP server started
LoRa init failed. Check your connections.

ricaun, 2019-04-24:
The right command should be `LoRaNow.setPins(18, 26);`
And by default, the esp32 board autoselect the pins CS 18 and DIO0 26.
You dont need to set the pins. Just run the example and should work.
=D

halukmy, 2019-04-24:
nope :( i tried all of them........ damn :D not works

halukmy, 2019-04-24:
in other framework i made it with

#define SCK     5    // GPIO5  -- SX1278's SCK
#define MISO    19   // GPIO19 -- SX1278's MISO
#define MOSI    27   // GPIO27 -- SX1278's MOSI
#define SS      18   // GPIO18 -- SX1278's CS
#define RST     14   // GPIO14 -- SX1278's RESET
#define DI0     26   // GPIO26 -- SX1278's IRQ(Interrupt Request)

    while (!Serial) ;
    SPI.begin(SCK, MISO, MOSI, SS);
    LoRa.setPins(SS, RST, DI0); // set CS, reset, IRQ pin

ricaun, 2019-04-24:
Are you selecting the right board ?! This pinout is from Heltec Wifi LoRa 32 (V2)

Try this code!

```
#define SCK 5 // GPIO5 -- SX1278's SCK
#define MISO 19 // GPIO19 -- SX1278's MISO
#define MOSI 27 // GPIO27 -- SX1278's MOSI
#define SS 18 // GPIO18 -- SX1278's CS
#define DI0 26 // GPIO26 -- SX1278's IRQ(Interrupt Request)

SPI.begin(SCK, MISO, MOSI, SS);
LoRaNow.setPins(SS, DI0);

if (!LoRaNow.begin()) {
    Serial.println("LoRa init failed. Check your connections.");
    while (true);
  }
```

halukmy, 2019-04-25:
https://www.aliexpress.com/item/TTGO-ESP32-SX1276-LoRa-868-915MHz-Bluetooth-WI-FI-Lora-Internet-Antenna-Development-Board-for-Arduino/32845370112.html?spm=a2g0s.9042311.0.0.2d154c4dkQMhnk

this is of board i try,

it works on old sample, but on your lib i got problem i dont know why

ricaun, 2019-04-25:
Did you manage to work with the last release of the [arduino-LoRa](https://github.com/ricaun/arduino-LoRa) you can try my resopitory is basic the same.

Try this example.
```
#define SCK 5 // GPIO5 -- SX1278's SCK
#define MISO 19 // GPIO19 -- SX1278's MISO
#define MOSI 27 // GPIO27 -- SX1278's MOSI
#define SS 18 // GPIO18 -- SX1278's CS
#define DI0 26 // GPIO26 -- SX1278's IRQ(Interrupt Request)

void setup() {
  Serial.begin(115200);
  SPI.begin(SCK, MISO, MOSI, SS);
  LoRa.setPins(SS, -1, DI0);
  while (!LoRaNow.begin()) {
    Serial.println("LoRa init failed. Check your connections.");
  }
  Serial.println("LoRa init OK!");
}

void loop() {

}
```

Should work, I realy dont know whats wrong =(

halukmy, 2019-04-25:
im going to try, do you have skype or insatnt message like whatsapp?

halukmy, 2019-04-25:
this one worked well btw :)

ricaun, 2019-04-25:
> 
> 
> im going to try, do you have skype or insatnt message like whatsapp?

Add me on Skype: luizcassettari

Try this example.
```
#include <LoRaNow.h>

#define SCK 5 // GPIO5 -- SX1278's SCK
#define MISO 19 // GPIO19 -- SX1278's MISO
#define MOSI 27 // GPIO27 -- SX1278's MOSI
#define SS 18 // GPIO18 -- SX1278's CS
#define DIO0 26 // GPIO26 -- SX1278's IRQ(Interrupt Request)

void setup() {
  Serial.begin(115200);
  Serial.println("LoRaNow SetPins");
  while (!Serial);
  SPI.begin(SCK, MISO, MOSI, SS);
  LoRaNow.setPins(SS, DIO0);
  while (!LoRaNow.begin()) {
    Serial.println("LoRa init failed. Check your connections.");
  }
  Serial.println("LoRa init OK!");
  LoRa.dumpRegisters(Serial);
}

void loop() {

}
```

ricaun, 2019-04-27:
Hello,
The new version has this function `LoRaNow.setPinsSPI(SCK, MISO, MOSI, SS, DIO0);`
Can help to make the code more clear.

closed 2019-04-27 by ricaun