ricaun/LoRaNow #8: Lora Nodes communication without using internet

turkone, 2019-10-03:
First of all i would like to thank you for all your studies. 
I want to make the lora nodes network(more than 5 nodes) those have their id and communicates with each other without using internet. I am using just LoRa Ra-02  SX1278 433 Mhz modules and Arduino Unos .Could you help me?

ricaun, 2019-10-12:
Hello,
You need to have a gateway to receive de messages if your gateway has internet is easier to control and send forward the nodes messages.
But you need to remember the node only receives a message after sends to the gateway.
The library was not designed to make a mesh network or make all the nodes talk with each other.
You need to have a gateway.
See yea!

turkone, 2019-10-12:
thanks for the answer .I got one more question.
I use two SX1278 LoRa Ra-02 modules for communication. I wonder how I can set IDs for these two modules to deny an interference with other LoRa modules which work at the same frequency.

I mean, is there a solution similar to pipe addresses of NRF24l01 communication. Thanks in advance.

ricaun, 2019-10-14:
I really don't know hot the pipe addresses of NRF24l01 works, if you need to set the ID of the node you can use this function `LoRaNow.setId(id);`, the library already deny any random packages how is not looks like the loranow protocol.

turkone, 2019-10-15:
That will be useful for my problem. Thanks a lot .

closed 2020-05-06 by turkone