ricaun/LoRaNow #13: Is LoRaNow.loop() necessary during sleeping

IoTopenTech, 2019-12-19:
Dear Ricaun,

I am learning to use your library.
I have a question about the LoRaNow.loop() method and the State Machine.
Can I sleep de uC in the onSleep callback for a long time (i.e. 15 minutes) without calling LoRaNow.loop()?

Besides I found this small typo in the API.
[image]
Kind regards from Madrid

IoTopenTech, 2019-12-19:
Another question.
When is this state enabled?
[image]

ricaun, 2019-12-21:
> I am learning to use your library.

👍 

> I have a question about the LoRaNow.loop() method and the State Machine.
> Can I sleep de uC in the onSleep callback for a long time (i.e. 15 minutes) without calling LoRaNow.loop()?

We the onSleep callback you can put the uC on sleep mode forever if you want.
The idea of the onSleep callback is to make sure the node sends the payload and waits for some response, then go to sleep.

> Besides I found this small typo in the API.

Opsss 😄 

> Another question.
> When is this state enabled?
> [image]

This state is not used, I create for the first version but never used.
I need to remove it.

IoTopenTech, 2020-01-03:
Thank you very much ricaun and Happy 2020...